Mechazo11/ros2_orb_slam3 #1: VSLAM with inertial using Intel D455?

mukundbala, 2024-01-29:
Awesome stuff here! The installation was a breeze. I was wondering how i can go about using ORBSLAM3 using my Intel D455 Camera. This camera has an IMU stream!

Mechazo11, 2024-01-29:
Hi @mukundbala thank you for your kind words. I guess you will need to start from thien94`s ORB-SLAM3 ROS1 package to see how he called the ORB-SLAM3 library with RGB+IMU stream. Then use my example of the monocular node I wrote, create another node and replicate the steps for the Intel D455 camera. Unfortunately, I don't have any plans for adding direct camera stream functionality any time soo due to severe time constraints.

Hope this helps.

With best,
@Mechazo11

closed 2024-01-29 by Mechazo11

mukundbala, 2024-01-29:
Once I get it working, I'd be happy to share my impl with you via a pull request of you'd like :) happy to contribute!

Mechazo11, 2024-01-29:
Sure, it will be my pleasure