Mechazo11/ros2_orb_slam3 #2: Colcon build failed "ModuleNotFoundError: No module named 'ament_package'"

TL-4319, 2024-03-01:
I am running into a problem performing colcon build for ORB-SLAM3. I followed the tutorial linked here https://www.youtube.com/watch?v=kQLPupPkQiU because I want to get ROS2 humble on a Jetson Orin which does not have Ubuntu 22.04 yet. The installation of ROS2 and the ZED wrapper is successful. At the end I have an underlay ros2-humble workspace underlay and zed-ws workspace overlay.

Then I proceed to install ros2_orb_slam3 by following the ReadMe. The only thing that I deviate from the instruction is to install Pangolin v0.6 instead of the newer v0.9 due to Ubuntu 20.04 does not have pybind11 2.6 which is what pangolin 0.9 needs. I have finished installing Pangolin.

What I have done for ros2_orb_slam3

- mkdir -p ./ros2_test/src
- cd ./ros2_test/src
- git clone https://github.com/Mechazo11/ros2_orb_slam3.git
- cd ..
- source ./zed-ws/install/setup.bash
- colcon build --symlink-install

which result in the following error
`Starting >>> ros2_orb_slam3
--- stderr: ros2_orb_slam3                          
Traceback (most recent call last):
  File "/home/user/Documents/ros2-humble/install/ament_cmake_core/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
CMake Error at /home/user/Documents/ros2-humble/install/ament_cmake_core/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(/usr/bin/python3
  /home/user/Documents/ros2-humble/install/ament_cmake_core/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  /home/user/Documents/ros2_test/build/ros2_orb_slam3/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  /home/user/Documents/ros2-humble/install/ament_cmake_core/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:41 (include)
  /home/user/Documents/ros2-humble/install/ament_cmake/share/ament_cmake/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)
  /home/user/Documents/ros2-humble/install/ament_cmake/share/ament_cmake/cmake/ament_cmakeConfig.cmake:41 (include)
  CMakeLists.txt:26 (find_package)

---
Failed   <<< ros2_orb_slam3 [3.67s, exited with code 1]

Summary: 0 packages finished [14.0s]
  1 package failed: ros2_orb_slam3
  1 package had stderr output: ros2_orb_slam3
`

Please let me know what might be a potential problem or mistake I made?

TL-4319, 2024-03-02:
For more information, the problem seems to happen when installing ROS2 from source instead of debian packages. I have another machine running native Ubuntu 22.04 with ROS2 installed from debian package. With that, colcon build successfully. However, I have to remove the ROS2 installation and build it from source due to the need for ros1_bridge. After building ros2-humble from source on the 22.04 machine and sourcing. Colcon build on orb-slam3 returns the same error message

Mechazo11, 2024-03-06:
Hi @TL-4319 apologies for the delay in getting back to you.

Unfortunately, I don't have an Jetson Orin board on hand to replicate your steps. But it sounds like, you may want to look into which version of ros2 humble you are downloading from the source to install in jetson orin since amnet_package must be present. I recall that from Humble already comes shipped with amnet_package.

But regarding not having Ubunutu 22.04 supported on Orin, have you looked at this https://forums.developer.nvidia.com/t/jetson-software-roadmap-for-2023/238889 ? If Jetpack 6.0 is out, you may want to switch to that to get ubuntu 22.04 natively in Jetson Orin.

I would suggest rather than building this package, try building a vanilla C++ package to see if the build passes to verify that amnet_packages is indeed installed correctly with your ROS2 installation.

I hope this helps.

Mechazo11, 2024-03-06:
@TL-4319 thank you for the kind word. I think lets close it for now since the issue is not related directly to this repository but more related to Nvidia ORIN platform itself

Good luck with your project.

closed 2024-03-06 by Mechazo11