Mechazo11/ros2_orb_slam3 #5: colcon build failed

Dmitrii173173, 2024-04-28:
/usr/bin/ld: /lib/libgdal.so.30: undefined reference to `curl_multi_wait@CURL_OPENSSL_4'
/usr/bin/ld: /lib/x86_64-linux-gnu/libnetcdf.so.19: undefined reference to `curl_global_cleanup@CURL_OPENSSL_4'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/mono_node_cpp.dir/build.make:231: mono_node_cpp] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:168: CMakeFiles/mono_node_cpp.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros2_orb_slam3 [0.58s, exited with code 2]

Summary: 0 packages finished [0.72s]
  1 package failed: ros2_orb_slam3
  1 package had stderr output: ros2_orb_slam3

can you help me please. have no idea why failed

Failed   <<< ros2_orb_slam3 [0.58s, exited with code 2]

Mechazo11, 2024-04-29:
Hi @Dmitrii173173 can you provide some information on the system you tried to build i.e. CPU, RAM specs, version of Ubuntu and version of ROS 2?

From the build error it appears you may be missing a library?
Maybe this might help

https://github.com/graphdeco-inria/gaussian-splatting/issues/152

Let me know if this solves the issue.

Mechazo11, 2024-05-04:
@Dmitrii173173  I am closing this issue for now. If require further assistance on this matter, feel free to open this issue back up.

closed 2024-05-04 by Mechazo11