Mechazo11/ros2_orb_slam3 #7: colcon build crashes because of memory overflow

TwooSix, 2024-05-26:
I compiling the project with 16GB of memory plus 16GB of swap, but it still crashes due to the memory and swap being fully occupied...

fong0802, 2024-05-26:
Same problem here, I have 8gb ram. Always happen around 4 minutes build time.

Mechazo11, 2024-05-27:
@fong0802 for your case, try not to have any chrome, mozilla like applications open. Increase your swap size to 24 Gb and see what happens.

@TwooSix can you post the crash message?

Mechazo11, 2024-05-27:
@TwooSix @fong0802 try to build only the ```ros2_orb_slam3``` package using this command

```colcon build --packages-select ros2_orb_slam3```

fong0802, 2024-05-28:
@Mechazo11 I setup 24 Gb more swap memory. I dint notice that i alrdy had 4 Gb swap. So total swap I setup is about 30Gb as shown by system monitor. Before I start to build, I directly switched off GUI.

I build the whole package with:
colcon build --symlink-installl

At the end, the package is sucessfully build with whole lot of warning, too much for me to keep track of.

Then, i try the example provided. 
The python node giving out error "no module named natsort"

After I installed the natsort module, it is working now.

If possible can you share briefly how you make it work in ROS2 humble? 

I am trying to implement ORB-SLAM on an Intel single board computer that is about 3 times more powerful than Raspberry pi 4. 
Since there are my own image processing algorithm running in it, I might need to port a light weight version myself. 
I am quite new to ROS. Any advice on that is appreciated. Thanks.

Mechazo11, 2024-05-28:
Hi @fong0802 

Awesome, I am glad you got to fix the issue.  

As I believe you are going to be using GitHub more frequently, consider picking up some of the Markdown syntaxes. This will help to differentiate code from text. An excellent tutorial is [here](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax)

Now let me try to answer some of your queries

*If possible can you share briefly how you make it work in ROS2 humble?* 

In the ```README.md``` file, I mentioned the ROS 1 port of **ORB SLAM3** made by [thien94](https://github.com/thien94/orb_slam3_ros) that inspired **ros2_orb_slam3**. The basic premise is simple, Dr. Campos`s ```orb-slam3``` [library](https://github.com/UZ-SLAMLab/ORB_SLAM3)  is essentially an ```internal shared``` **library** of the ```ros2_orb_slam3``` package. You can see this arrangement in ```CmakeLists.txt``` file, more specifically at ```line 72```. Whenever you update any files in ORB SLAM3 say, you wanted to add a file called ```CommonStructs.h``` to collect some ```Struct``` objects used throughout, you would do the following

```
add_library(orb_slam3_lib SHARED
  orb_slam3/src/System.cc
  orb_slam3/src/Tracking.cc
  orb_slam3/src/LocalMapping.cc
  orb_slam3/src/LoopClosing.cc
  orb_slam3/src/ORBextractor.cc
  orb_slam3/src/ORBmatcher.cc
  orb_slam3/src/FrameDrawer.cc
  orb_slam3/src/Converter.cc
  orb_slam3/src/MapPoint.cc
  orb_slam3/src/KeyFrame.cc
  orb_slam3/src/Atlas.cc
  orb_slam3/src/Map.cc
  orb_slam3/src/MapDrawer.cc
  orb_slam3/src/Optimizer.cc
  orb_slam3/src/Frame.cc
  orb_slam3/src/KeyFrameDatabase.cc
  orb_slam3/src/Sim3Solver.cc
  orb_slam3/src/Viewer.cc
  orb_slam3/src/ImuTypes.cc
  orb_slam3/src/G2oTypes.cc
  orb_slam3/src/CameraModels/Pinhole.cpp
  orb_slam3/src/CameraModels/KannalaBrandt8.cpp
  orb_slam3/src/OptimizableTypes.cpp
  orb_slam3/src/MLPnPsolver.cpp
  orb_slam3/src/GeometricTools.cc
  orb_slam3/src/TwoViewReconstruction.cc
  orb_slam3/src/Config.cc
  orb_slam3/src/Settings.cc
  orb_slam3/include/System.h
  orb_slam3/include/Tracking.h
  orb_slam3/include/LocalMapping.h
  orb_slam3/include/LoopClosing.h
  orb_slam3/include/ORBextractor.h
  orb_slam3/include/ORBmatcher.h
  orb_slam3/include/FrameDrawer.h
  orb_slam3/include/Converter.h
  orb_slam3/include/MapPoint.h
  orb_slam3/include/KeyFrame.h
  orb_slam3/include/Atlas.h
  orb_slam3/include/Map.h
  orb_slam3/include/MapDrawer.h
  orb_slam3/include/Optimizer.h
  orb_slam3/include/Frame.h
  orb_slam3/include/KeyFrameDatabase.h
  orb_slam3/include/Sim3Solver.h
  orb_slam3/include/Viewer.h
  orb_slam3/include/ImuTypes.h
  orb_slam3/include/G2oTypes.h
  orb_slam3/include/CameraModels/GeometricCamera.h
  orb_slam3/include/CameraModels/Pinhole.h
  orb_slam3/include/CameraModels/KannalaBrandt8.h
  orb_slam3/include/OptimizableTypes.h
  orb_slam3/include/MLPnPsolver.h
  orb_slam3/include/GeometricTools.h
  orb_slam3/include/TwoViewReconstruction.h
  orb_slam3/include/SerializationUtils.h
  orb_slam3/include/Config.h
  orb_slam3/include/Settings.h
  orb_slam3/include/CommonStructs.h
)

``` 

The rest is run-of-the mill ```Cmake``` rules of adding library, creating executables and linking executables with internal shared and external libraries. Check my ```CmakeLists.txt```, that would be your ideal starting point.

* I am trying to implement ORB-SLAM on an Intel single board computer that is about 3 times more powerful than Raspberry pi 4.*

ORB-SLAM3 is a CPU heavy library which in our experience did not run well with Raspberry pi 4 like SoCs. I would strongly recommend going with at least the new Orin Nano board, as it may then be possible for you to run your image processing algorithms on GPU, use the python node to then transfer those results to ORB SLAM3 leaving most of the CPU computation powers for running ORB SLAM3. 

Hope these suggestions helped. Good luck with your work.

With best,
@Mechazo11

fong0802, 2024-05-28:
@Mechazo11 Thanks for your help!! These give me a good starting point. 

Good luck with your publications too.