Mechazo11/ros2_orb_slam3 #10: Compilation problems

OdedHorowits, 2024-06-17:
Hi
I tried to `colcon build` this repo inside a docker on my laptop - i5 with 8GB RAM, the laptop kept freezing.
I limited the resources to 3GB and 4 processors then got error `c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.` which seems to be connected to insufficient resources.
So I changed to a gaming pc with 30GB and 32 cores and limited the docker container to 'just' 12GB and 16 cores.
15 minutes already, still on 87%.

Should it take so much time and so many resources??

OdedHorowits, 2024-06-17:
solved that by not limiting the resources at all

closed 2024-06-17 by OdedHorowits

Mechazo11, 2024-06-27:
Hi @OdedHorowits great, happy to hear that you were able to solve this problem. Good luck with your project.