Mechazo11/ros2_orb_slam3 #14: Using Real Time

CharlieRusso001, 2024-07-09:
Hello,
I was looking to use this package in order to stream image and camera position real time and from the REAME there isn't a documentation for this repository.  I was wondering if you all had any documentation on this repository so I could use it in order to stream images and poses through ros2 humble with a realsense camera, and if so could you provide it?

I was intending to use this wrapper for NeRF with this repo:
https://github.com/javieryu/nerf_bridge

Also, there seems to be some camera pose artifacts on the provided monocular example, however they do not seem to create any data.  Why does this happen? ( The arc looking structure above the red square)
[image]

meard, 2024-07-09:
Hello @CharlieRusso001 
If I understand correctly, you are looking for reference to use live image streaming with ORB SLAM3 right?  
You can try my [repository](https://github.com/meard/ros2_orb_slam3) where I have implemented image streaming from Gazebo to ORB SLAM3. Little research and modification to the code might help you achieve what you might require.

Mechazo11, 2024-07-09:
Hi @CharlieRusso001 

This repo is a ```bare-bones``` ROS 2 implementation of ORB SLAM 3 V 1.0 library which originally was not implemented to work in real-time. Hence there is no such documentations to provide you in this repo. I suggest you being with @meard repository. Ultimately this was my goal with this repo, to provide developers a ```bare-bones``` ROS 2 port of ORB-SLAM3.