Mechazo11/ros2_orb_slam3 #21: colcon build is failing with 1 package built and the second one failing due to many warnings(which I doubt is

VASISHTAKODUMAGULLA, 2025-01-03:
when I am trying to build the

VASISHTAKODUMAGULLA, 2025-01-03:
The Main issue is I don't understand what is the error here since everything is only a warning.

Mechazo11, 2025-01-03:
Hi @VASISHTAKODUMAGULLA 
Can you please provide some additional information i.e. OS, RAM, CPU you are building this on?
Secondly I don't think the build failed, it did build the package correctly. Can you run the sample monocular image example?

VASISHTAKODUMAGULLA, 2025-01-03:
Hello again,
Sure, The OS is Ubunutu 22.04 (Jammy jelly fish) and ROS 2 Humble hawksbull with Python 3.10.
System details: 
[image]

VASISHTAKODUMAGULLA, 2025-01-04:
Hi Kamal,
Yes it did build partially and I did run the sample monocular image node it failed to run

Mechazo11, 2025-01-04:
Can you post some screenshots of what type of failure it is reporting? If you are planning on copying the console outputs after an error has been thrown, consider formatting them in github's markdown format. A sample is shown below

```bash
[Error] Some message
```

This guide may be of some use: https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

VASISHTAKODUMAGULLA, 2025-01-04:
[image]

Here is the error response from the terminal

Mechazo11, 2025-01-04:
@VASISHTAKODUMAGULLA please go over the updated README file in my repo for a new set of instructions in installing and testing this repo.

The problem is your ```vision_opencv``` metapackage that contains ```cv_bridge``` might have been corrupted or there is a linking issue to that library. I would suggest doing the following

* Step 1: Completely delete ros2 humble and reinstall it
* Step 2: Delete the entire ```ros2_test``` workspace
* Step 3: Create the ```ros2_test``` workspace and build the orb slam 3 package following the new instructions.

Hopefully this will solve the runtime error you are seeing. For now I will go ahead and close this ticket since the problem is not with building the `ros2_orb_slam3` package.

VASISHTAKODUMAGULLA, 2025-01-07:
Hello Kamal, Thank you for your help. I have reinstalled the ros2 humble and tried as you have suggested and it did work though there was some stderr while building and in the end ORB slam3 run as expected but I am trying to implement stereo inertial mode with a different stereo dataset on your setup to obtain the output as position and orientation in a log file. do you have any suggestions how to obtain them?
I would really appreciate if you could suggest me something..

Mechazo11, 2025-01-07:
Hi @VASISHTAKODUMAGULLA glad to hear it worked out. ```stderr``` in my opinion are `warnigns` then actual errors so you can pretty much disregard them.

Regarding where to start integrating stereo intertial mode I would suggest starting from here: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/Examples/Stereo-Inertial/stereo_inertial_euroc.cc

but remember you will have to update the python node to pass two images to the C++ node and also the IMU data too. Unfortunately, I cannot help any further beyond this suggestion since I work almost exclusively with RGB-D and Monocular data.

Good luck.

Mechazo11, 2025-01-27:
Hi @VASISHTAKODUMAGULLA yes, you have to make changes in the C++ side also. 

Check the `stereo_intertial_euroc.cc` file here: https://github.com/UZ-SLAMLab/ORB_SLAM3/tree/master/Examples/Stereo-Inertial and the Cmake file here https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/master/CMakeLists.txt. Both are from the original non-ros version of ORB-SLAM3. 

The basic premise is very similar to my monocular node, you need to pass timestep, image and imu data to the appropriate method withint the ORB-SLAM3`s ```System``` class.

For example, in the ```common.cc``` file found in ```ros2_orb_slam3/src``, the ```MonocularMode::Img_callback()``` method essentially calls the following

```Sophus::SE3f Tcw = pAgent->TrackMonocular(cv_ptr->image, timeStep); ````. 

For your case this would probably end up being something like

```c++
// Pass the images to the SLAM system
pAgent->TrackStereo(imLeft,imRight,tframe,vImuMeas);
````

where ```TrackStereo``` is the method responsible for activating ```Stereo``` mode in ```ORB-SLAM3```. Check the ```stereo_intertial_euroc.cc``` for more details.

Hope this answers your question.

Mechazo11, 2025-02-03:
@VASISHTAKODUMAGULLA closing since implementing stereo tracking may be a future version upgrade.

closed 2025-02-03 by Mechazo11

VASISHTAKODUMAGULLA, 2025-02-10:
Hello Kamal,
I have been playing with mono setup currently to grab a deeper understanding and I am pondering if there is any ground truth values for the sample Euroc data set that you have used? I do see that there is a ground truth values in the file saved but these values haves different time stamps and are also more than what we have in the test dataset.

Mechazo11, 2025-02-10:
@VASISHTAKODUMAGULLA yes check ETH ASL's EuRoC MAV dataset. It is very well known in SLAM community. 

I would also suggest opening a new ticket in future rather than commenting on a close ticket on topics that aren't related to the original ticket.