Mechazo11/ros2_orb_slam3 #23: build fails because <cv_bridge/cv_bridge.h> and <image_transport/image_transport.h> are not found

SpaceSoarer90, 2025-02-02:
ROS2 Version: ros2 jazzy
Ubuntu Version: Ubuntu 24.04.1 LTS (noble)

[image]
[image]

==================================================================

changing "cv_bridge.h" and "image_transport.h" to "cv_bridge.hpp" and "image_transport.hpp" completes the build

[image]

==================================================================

is this because i'm using ros2 jazzy and not humble?

Mechazo11, 2025-02-03:
Hi @SpaceSoarer90 thank you for opening the two issues. 

I didn't experience this issue with my test with ros2 jazzy. I will investigate it and let you know soon. In the meantime have you gotten the system to build and run in your ROS 2 Jazzy? If modifying the `.h` files to `.hpp` solves this and Issue #24, I will go ahead and open a separate branch and then invite you to make a pull request on that branch.

SpaceSoarer90, 2025-02-04:
yes, i was able to build the system properly after editing the mentioned common.hpp file and CMakeLists.txt file ( #24 ).

i was also able to run the monocular example provided in the main page of this repo.
[image]

perhaps, the build error has something to do with my OpenCV version, which is currently 4.6.0.
```
ros2-devel% python3 -c "import cv2; print(cv2.__version__)"
4.6.0
```

Mechazo11, 2025-03-03:
To future users, I am not sure whether this issue was a one off problem. See #26 for a discussion where an user was able to get this package built in Jetson Orin Nano without needing to do the changes suggested here. For now I am closing this issue. @SpaceSoarer90  feel free to opening it back up if required

closed 2025-03-03 by Mechazo11