Mechazo11/ros2_orb_slam3 #25: ORB SLAM3 Evaluation against the ground truth

VASISHTAKODUMAGULLA, 2025-02-11:
Hello Kamal,
I know this is out of your Job but since I have been implementing your work wanted to ask when I tried downloading the dataset from Euroc mav VI dataset at http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/machine_hall/MH_05_difficult/MH_05_difficult.zip I see that after downloading, the Ground truth timestamps some how doesn't include all the images timestamps provided in the data set I dont understand what to do in this case. 
For Instance, I picked a Image timestamp and searched for it in the ground truth estimate timestamps and I dont find it there for instance thsi one 1403638627677829376,1403638627677829376.png and also the last time frame in the data set 1403638631677829376,1403638631677829376.png and these are just some of them and I believe there are more like that.

Any suggestions how to proceed? 
Regards,
Vasista

Mechazo11, 2025-02-11:
Hi @VASISHTAKODUMAGULLA 

Firstly, ORB-SLAM3 is not my own work. It came out of Dr. Campo's Ph.D. work. I just created a wrapper to make it easier for developers to use it in ROS 2.

Regarding your problem, you may want to use `Path` and `glob` libraries together to sample those timestamps for which ground-truth data is available but I think there may be very very small difference between the timestamps that names the images to timestamps when a sample point was recorded from the MoCAP system. To the best of my knowledge, I don't recall the EuRoC MAV paper stating they were 100% accurate in achieving temporal alignment between recording raw data and ground-truth positions.

Alternatively, first just perform inference with all the available images and then use Micheal Gruppo's [evo](https://github.com/MichaelGrupp/evo) tool to perform map alignment and ATE calculations between estimated pose and ground-truth pose. I used the latter for my IEEE AIM paper and I strongly suggest you do that. A modified version of ros2 ORB-SLAM3 that enables recording estimated poses can be found [here](https://github.com/RKinDLab/ros2_psd_pcb_reloc).

I hope this helps.

Mechazo11, 2025-02-21:
@VASISHTAKODUMAGULLA  I am closing this issue. Feel free to opening it back up at a later time.

closed 2025-02-21 by Mechazo11