Mechazo11/ros2_orb_slam3 #26: Build Issues

snehabanda30, 2025-02-20:
I am trying to build the orb_slam_3 algorithm on my Ubuntu 22.04, using the ROS:Humble workspace on my Jetson Orin Nano. However, I am running into issues with the build. This is the docker file, which installs pangolin. I create the ros_test workspace, disregard that it creates ros_ws in the file. However, I am still running into issues.  I have uploaded my dockerfile. 

 

[image] 

pangolin is installed 
[image] 

[Dockerfile (1).txt](https://github.com/user-attachments/files/18896457/Dockerfile.1.txt)

Mechazo11, 2025-02-21:
Hi @snehabanda30,

Thank you for reaching out. Unfortunately, I don't have a Orin Nano to help you troubleshoot this issue directly. Furthermore, a photo of the console window often is insufficient to analyze build errors.

Having said that, to figure out a solution, I would need the following information

1. Jetson Orin Nano (and its software upgraded Super version) supports Ubuntu 22.04 and ROS 2 Humble natively. Why are you trying a Docker build without doing a native build first?

2. Orin Nano super has 8GB RAM. ORB-SLAM3 in my own and a few other users of this repo observed that build failure often happens due to insufficient RAM. There is a strong possibility this is the case too. A workaround is to assign 16GB of `SWAP` space that emulates `RAM` on the disk. See this article for more info: https://www.forecr.io/blogs/programming/how-to-increase-swap-space-on-jetson-modules  

3. Can you copy the error message from the console window and format it in Github Markdown so that I can see the entire error log. Please refer to this article: https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

Best,
@Mechazo11

jesseknuckles, 2025-02-21:
So I tried it locally instead of docker, and I was able to build pangolin without any issues, but I am running into issues with the download. 

```
/usr/bin/ld: /home/c9-jetson1/ros2_test/src/ros2_orb_slam3/orb_slam3/Thirdparty/DBoW2/lib/libDBoW2.so: error adding symbols: file in wrong format
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [CMakeFiles/orb_slam3_lib.dir/build.make:645: liborb_slam3_lib.so.] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/orb_slam3_lib.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros2_orb_slam3 [5min 26s, exited with code 2]
                                      
Summary: 0 packages finished [5min 27s]
  1 package failed: ros2_orb_slam3
  1 package had stderr output: ros2_orb_slam3

```

Mechazo11, 2025-02-22:
@jesseknuckles the error suggests the `libDBow2.so` object was built for a CPU architecture not supported by Jetson Orin Nano. Can you try this command (ChatGPT suggested)

```bash
file /home/c9-jetson1/ros2_test/src/ros2_orb_slam3/orb_slam3/Thirdparty/DBoW2/lib/libDBoW2.so
```

In my opinion, you may have to rebuild the DBoW2 [library](https://github.com/dorian3d/DBoW2) from scratch inside Orin Nano, then copy over the `libDBoW2.so` file with the newer version inside [ThirdParty](https://github.com/Mechazo11/ros2_orb_slam3/tree/main/orb_slam3/Thirdparty)/DBoW2 and rerun the build command to build the ros2_orb_slam3 package.

Hope this helps.
Best,
@Mechazo11

snehabanda30, 2025-03-03:
Hey @Mechazo11 Thank you very much for the help so far. Now, we have this repo and the ros realsense wrapper built in the same ros workspace. we ran the ros2 launch realsense2_camera rs_launch.py align_depth:=true command to launch the camera/camera node to enable us to interface with the RealSense via ros. Then we changed the mono-driver-node.py file so that it interacts with this /camera/camera node, and publishes sensor_msgs/msg/Image messages to the /camera/image_raw topic. 

Then I run ros2 run ros2_orb_slam3 mono_node_cpp --ros-args -p node_name_arg:=mono_slam_cpp which responds with "Waiting to finish handshake ..."

Then I run ros2 run ros2_orb_slam3 mono_driver_node.py. But the mono_node_cpp executable still stays on "Waiting to finish handshake"  When I inspect the topic info of /camera/image_raw, it shows a publisher (from the mono_driver_node.py script) but no subscribers. I tried to use the --remap flag on the mono_node_cpp executable, but I am unsure what ros topics that executable is trying to read from. 

If I just change the python script to publish to the topics that mono_node_cpp expects to read from will this be the solution I am looking for? And if so, what are those topics, because I am unsure how to figure out what topics the mono_node_cpp executable is trying to read from.

Please let me know if I am on the right track, and if so, what the topic is called that I should publish to from the python script.

Mechazo11, 2025-03-03:
Hi @snehabanda30 

I think you are in the right track. 

But recall, mono_driver_node.py, in my vainlla repo, is reading from a static file. You will have to make changes so that both the `rgb` and `depth` data (which should be two separate topics) are first read by `mono_driver_node.py` first, they in turn then passes those information to some subscriber to the cpp node.

To make sure this data flow actually works, once you fire up the `cpp` node, you then activate the `py` node which then calls the ```handshake_with_cpp()``` function which will then complete the handshake with `cpp` node and in then in the subsequent timesteps pass data.

@snehabanda30 I think this will be more prudent if you can confirm whether you were able to successful build this package in Jetson Orin Nano. If so, in the next reply,  can you describe how you had done and lets close this ticket. We can open another ticket to discuss a solution to this handshake issue.

AbleToCode, 2025-03-04:
Hi, I'm trying to build the package with colcon build --symlink-install, but the build fails during the linking phase with the following errors:

[image]

I'm running this on WSL Ubuntu 20.04.

The linker warning indicates that it needs libopencv_core.so.4.5d, but the README states that OpenCV 4.2 is sufficient. Should I update my OpenCV version, or is there another issue?

snehabanda30, 2025-03-07:
Initially, we begin  by checking each library's os system. 
``` 
file /home/c9-jetson1/ros2_test/src/ros2_orb_slam3/orb_slam3/Thirdparty/DBoW2/lib/libDBoW2.so
``` 
It was built for x86, and then we rebuilt the libraries for aarch64. 
Afterwards, we built each library. We followed the instructions, as written.  Then, we ran colcon build --cmake-args -DCMAKE_BUILD_PARALLEL_LEVEL=6., using 6 cores.

closed 2025-03-07 by snehabanda30

Mechazo11, 2025-03-07:
Thanks @snehabanda30 it is really impressive that this package now works in jetson nano, I will update the readme to point to this ticket for future users who would be working to implement this package in Orin Nano boards.