Mechazo11/ros2_orb_slam3 #38: Can you provide descriptions of configuration for custom images?

Algtronic, 2025-06-24:
It's not clear how to place custom imageset in image generation node

Mechazo11, 2025-06-30:
No worries @Algtronic. Lets see the first few lines of the MonoDriver(Node) class in `mono_driver_node.py`

```python
class MonoDriver(Node):
    def __init__(self, node_name = "mono_py_node"):
        super().__init__(node_name) # Initializes the rclpy.Node class. It expects the name of the node

        # Initialize parameters to be passed from the command line (or launch file)
        self.declare_parameter("settings_name","EuRoC")
        self.declare_parameter("image_seq","NULL")

        #* Parse values sent by command line
        self.settings_name = str(self.get_parameter('settings_name').value) 
        self.image_seq = str(self.get_parameter('image_seq').value)

        # DEBUG
        print(f"-------------- Received parameters --------------------------\n")
        print(f"self.settings_name: {self.settings_name}")
        print(f"self.image_seq: {self.image_seq}")
        print()

        # Global path definitions
        self.home_dir = str(Path.home()) + "/ros2_test/src/ros2_orb_slam3" #! Change this to match path to your workspace
        self.parent_dir = "TEST_DATASET" #! Change or provide path to the parent directory where data for all image sequences are stored
        self.image_sequence_dir = self.home_dir + "/" + self.parent_dir + "/" + self.image_seq # Full path to the image sequence folder

        print(f"self.image_sequence_dir: {self.image_sequence_dir}\n")
```

Observe that you can set the settings name to `EuRoC` that points to `EuRoC.yaml` file, the name of the image sequence `image_seq`. As I had mentioned in `README.md`, the python node is very simple in nature, and is meant to be a minimal example.

Hope this answers your question.

Mechazo11, 2025-06-30:
My pleasure. In my opinion, the naming schema kinda works, see this folder
https://github.com/Mechazo11/ros2_orb_slam3/tree/main/orb_slam3/config/Monocular

These are all the camera configuration tied to each dataset you can use. The name `EuRoC` comes from the ASL Lab, see here https://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets