Mechazo11/ros2_orb_slam3 #41: how to use a webcam with this

MARS-PESU, 2025-08-12:
Hey, so I had this confusion even with the original orb_slam3 repo. I am not able to understand how do i use a normal, monocular rgb webcam to stream video and then use this orb_slam to process the frames and get output.
Please do help me with that.
thanks in advance

MARS-PESU, 2025-08-14:
> Hi @MARS-PESU <https://github.com/MARS-PESU>
> You will need to implement a cv2.VideoCapture object to pass an RGB image
> frame with some sort of id into mono_driver_node:run_py_node method. See
> the following example from Claude as a guide
>
> import cv2import time
> # ****************************************************************************************def run_py_node(self, idx, frame):
>     """        Master function that sends the RGB image message to the CPP node    """
>     # Initialize work variables
>     img_msg = None # sensor_msgs image object
>     timestep = time.time()  # Use current timestamp
>     self.frame_id = self.frame_id + 1
>
>     # Convert captured frame directly instead of reading from file
>     img_msg = self.br.cv2_to_imgmsg(frame, encoding="passthrough")
>     timestep_msg = Float64()
>     timestep_msg.data = timestep
>
>     # Publish RGB image and timestep
>     try:
>         self.publish_timestep_msg_.publish(timestep_msg)
>         self.publish_img_msg_.publish(img_msg)
>     except CvBridgeError as e:
>         print(e)# ****************************************************************************************
>         # main functiondef main(args = None):
> [... 37 lines cut ...]
>     n.destroy_node()
>     rclpy.shutdown()
>
> Let me know if this guideline works, I would really welcome a PR to this
> ticket that adds this functionality into the project.

Is there a way to directly get feedback via ros messages, that would help
me cause I wanted to use camera for other tasks too.

MARS-PESU, 2025-08-14:
> Hi [@MARS-PESU](https://github.com/MARS-PESU) You will need to implement a `cv2.VideoCapture` object to pass an RGB image frame with some sort of id into `mono_driver_node:run_py_node` method. See [...]

> Let me know if this guideline works, I would really welcome a PR to this ticket that adds this functionality into the project.

thanks
I think I understand,
i will try this out this weekend, please help me out if I go wrong
I willl share the steps if it works out

Mechazo11, 2025-08-16:
Hi @MARS-PESU, I am going to close this ticket off for the time being. Feel free to opening it back up and I would greatly appreciate the PR to merge this feature if you would like to contribute.

Best,
@Mechazo11

closed 2025-08-16 by Mechazo11